cpc/openasip #5: pckage imem_mau not generated during processor generation

boehmerst, 2015-12-26:
Hello all,

it seems like the vhdl file defining the package imem_mau ist not generated during the processor generation. The same happens for the corresponding verilog include file during verilog code generation.

Regards,
Stephan

boehmerst, 2015-12-26:
The mentioned package gets generated when using the c2vhdl tool. Maybe I miss something when using the ProcGen GUI?

Stephan

postoroniy, 2015-12-27:
Hi Stephan,
From manual
```text
Generate instruction memory bit image using Program Image Generator. Finally, to get our shiny
new processor some bits to chew on, we use generatebits to create instruction memory and data memory
images:
generatebits -d -w 4 -p crc_with_custom_op.tpef -x proge-output custom.adf
Flag −d creates data images, −w4 set data memory width 4 MAUs (Minimum Addressable Units, bytes
in this case), −p defines the program, −x defines the HDL directory, and adf is given without a flag.
```
I guess that you should notice some difference between c2vhdl and your custom command

boehmerst, 2015-12-27:
Hi,

tnx for your feedback. Indeed the generatebits command made the difference! Unfortunatelly I was not aware of bitfile generation was involved in VHDL generation.

closed 2015-12-27 by boehmerst