cpc/openasip #8: Platform integrator broken?

alekob, 2016-04-01:
I've tried to execute the platform integrator in TCE 1.13 for tce_tutorials/fpga_stdout, as described in the user manual chapter 3.8.2ff:

**#>generateprocessor -i test.idf -o proge-out -g Stratix2DSP -d onchip -f onchip -p lightweight_print.tpef -e hello_tta test.adf

TTA core doesn't have compatible memory interface:
MemoryGenerator does not have port data_in**

It seems that the processor itself is generated correctly but there is no platform description!?

linehill, 2016-04-04:
Hi alekob,

I could not reproduce the described issue. However, I used tutorial's predefined IDF (preselected.idf) - when executing the command - instead of selecting implementations using ProDe's _Processor Implementation..._-dialog. 

How did you define the IDF in the tutorial? 

My best guess for now is that the LSU function unit has incompatible implementation selected in the IDF, which should be _fu_lsu_with_bytemask_always_3_ (ID: 2) from _stratixII.hdb_ for the tutorial.

alekob, 2016-04-12:
You are right, I'm sorry! Indeed I've used the LSU from asic_130nm_1.5V.hdb instead of the Stratix HDL

closed 2016-04-12 by alekob