cpc/openasip #43: seg. fault for generatebits

alekob, 2016-10-27:
generatebits fails with an segmentation fault when the compressor "InstructionDictionary" is used. Please unpack the attached archive and do a "make idict" 
[generatebits.zip](https://github.com/cpc/tce/files/556738/generatebits.zip)

alekob, 2016-11-07:
Just for clarification: this issue is not related to the proxim debugger window issue. Its still exists in the latest version!

alexhrv, 2016-11-08:
Thank you, we are aware of this. Seems like it has something to do with **ensure_programmability**-switch (setting it to **no** doesn't create segfault). The issue is still under investigation

linehill, 2016-11-08:
The source of the issue has been located and the fix is commited soon after I have tested it.

commit ad3eb196 (2016-11-08): Fix for seg. fault for generatebits

Fixes #43
closed 2016-11-08 by linehill