cpc/openasip #49: Unexpected jump address

alekob, 2017-03-02:
Using TCE 1.14 / LLVM 3.9.x a jump to an unexpected address is performed when reading the status of an volatile register. The code looks as follows:

typedef union {
	uint8_t all;

	struct {
		uint8_t reserved  : 3;
		uint8_t t_rh_done : 1;
		uint8_t rp_done   : 1;
		uint8_t start_dsp : 1;
		uint8_t pwm_run   : 1;
		uint8_t ready     : 1;
	} field;
} dsp_ctrl_t;

typedef struct {
	....
	dsp_ctrl_t    dsp_ctrl;
} CFG_t;

volatile CFG_t * const p_cfg = (CFG_t *) 0x100;

int main(void) {
	init_cfg();
	while (1) {
		if (p_cfg->dsp_ctrl.field.start_dsp) {
                ...
			p_cfg->dsp_ctrl.field.ready = 1;
		}
}

In the HW, dsp_ctrl.start_dsp is set to 0 when dsp_ctrl.ready = 1; therefore the expected behavior is that dsp_ctrl.ready is set to 1 only once after dsp_ctrl.start_dsp == 1. The HW behavior is working as expected (checked by simultion) but the SW still sets the ready flag to 1.

I've checked the code generated by tcecc and found, that the loop is not executed correctly (IMHO):

...
RF_8x32.5 -> LSU.in1t.ldqu ;       # @473
... ;                                                # @474
... ;                                                # @475
LSU.out1 -> INT_ALU.in2 ;           # @476
4 -> INT_ALU.in1t.and ;               # @477
INT_ALU.out1 -> INT_ALU.in2 ;   # @478
0 -> INT_ALU.in1t.eq ;                 # @479
INT_ALU.out1 -> BOOL.0 ;          # @480
?BOOL.0 476 -> gcu.pc.jump ;    # @481
...

Line 481 executed the loop to the point where the output of the last LD operation is read (476) instead of jumping to the point where the LD op is executed (473). Therefore the value of dsp_ctrl.start_dsp is not updated.

hkultala, 2017-03-03:
The load is in the delay slot of the jump, the delay slot instructions of the jump are always executed before the jump is actually taken.

We compiled similar code here and the code seemed to be correct, though LLVM had reorganized the  control structures quite heavily.

The problem here seems to be that the bits in bitfield in big-endian TCE are in big-endian format while you are propably assuming the bits come in in little-endian format. So in big-endian TCE the ready flag is the lowest bit, start_dsp flag is the third lowest bit.

alekob, 2017-03-06:
Little / bog endian is not the issue because writing to a certain bit field in an simple test case works (checked by simulation)

Anyways, the problem seems to be related to the struct which describes the HW registers. It is declared as "volatile" but the register is read only once while the program waits until the register is set to a certain value. Is there a special trick to make volatile registers work with tcecc/clang? Currently it's declared as follows:

typedef volatile struct {...} REGS_t;
REGS_t * const p_regs = (REGS_t *) 0x0;

Typed with thumbs. Please excuse typos.

pjaaskel, 2017-05-12:
Please provide a full test case.

alekob, 2017-06-23:
I'm working on a test case, by now the error happens the following situation:
- a single LSU (eg 512 byte) is used to access both data memory and registers. The lower half of the address range is assigned to the memory (eg 256 byte). This is done in HW
- the stack pointer is initialized to the upper boundary of the data memory using --init-sp 0x100
- in the case that the amount of data stored in the memory exceeds or is close to the size of the memory access to a register (mapped to addresses > 0x100) fail

It seem that the compiler doesn't honor the memory limit defined by the initial stack pointer setting. This results in a memory overflow which causes overwriting register contents

pjaaskel, 2019-03-26:
If I understood this issue correctly, you'd like the compiler to inject checks for stack pointer overflow and do "something" if there is overflow? This is unlikely to happen due to the runtime overheads it adds.

closed 2019-03-26 by pjaaskel