cpc/openasip #53: User manual: LaTeX converting --prefix to -prefix

pjaaskel, 2017-05-12:
The tutorial suggests the following command.

cp $(tce-config -prefix)/share/tce/data/mach/minimal.adf start.adf

alexhrv, 2017-05-12:
Could not spot the problem in the latest pdf file, both dashes are there:

[image]

pjaaskel, 2017-05-12:
Seems to be a problem in the HTML version. http://tce.cs.tut.fi/user_manual/TCE/node10.html#SECTION00411000000000000000

linehill, 2017-05-12:
Replacing '--'s with '{--}' in the latex doc seems to work for PDF output but not for HTML output (with latex2html).

alexhrv, 2017-05-16:
Replacing **\shellcmd** and surrounding the commands with **\begin{verbatim}** and **\end{verbatim}** in the latex doc file seems to fix this issue for both pdf and html without the need to escape the '--' characters.

pjaaskel, 2017-05-16:
OK please commit.

closed 2019-03-19 by alexhrv