cpc/openasip #56: Does this library compile C or Llvm?

vitiral, 2017-06-16:
In the README it states

> The toolset provides a complete co-design flow from C programs down to synthesizable VHDL and parallel program binaries. 

However on your [releases page](

pjaaskel, 2017-06-16:
If one uses LLVM internally it doesn't mean all the languages in the world that might have frontends that compile to LLVM IR work, are tested, or are 'supported'. E.g. C is more than the language: standard libraries, runtime support etc. C99 is the main language standard we use and support, OpenCL C is the 2nd. Many constructs of C++ work too, but not the whole stdlib etc.

closed 2017-06-16 by pjaaskel

vitiral, 2017-06-16:
Thankyou for responding.

I would just like to confirm that you do use the LLVM IR to create the HDL right? I understand that you support/test mainly C.

I was curious whether it could be theoretically possible to support [rust](https://www.rust-lang.org/en-US/) in the future. It is as low-level as C (no runtime) and also compiles to LLVM. I believe they share a lot in common once optimized to LLVM IR.

pjaaskel, 2017-06-16:
TCE is a hardware-software co-design toolset for design and programming of application-specific co-processors. The code for the customized cp-processors is produced with an LLVM-based compiler and the produced code is used to figure out what are good resources to add to the processor's resources etc. This is typically an iterative designer-directed/semi-automated/automated process.

Thus, yes LLVM IR is kind of used in the process when tailoring the architecture for which the HDL is generated. I don't see why Rust could not be supported relatively easily, if someone wishes to do so.