cpc/openasip #73: TCE fails to build with GCC 8.x

franz, 2018-08-14:
When building ``CopyingDelaySlotFiller.cc`` it fails with:

```
In file included from /usr/include/c++/8.1.1/set:60,
                 from CopyingDelaySlotFiller.cc:33:
/usr/include/c++/8.1.1/bits/stl_tree.h: In instantiation of ‘class std::_Rb_tree<boost::shared_ptr<ProgramOperation>, boost::shared_ptr<ProgramOperation>, std::_Identity<boost::shared_ptr<ProgramOperation> >, ProgramOperation::Comparator, std::allocator<boost::shared_ptr<ProgramOperation> > >’:
/usr/include/c++/8.1.1/bits/stl_set.h:133:17:   required from ‘class std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator>’
/usr/include/c++/8.1.1/bits/stl_pair.h:204:11:   required from ‘struct std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> >’
/usr/include/c++/8.1.1/ext/aligned_buffer.h:54:65:   required from ‘struct __gnu_cxx::__aligned_membuf<std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> > >’
/usr/include/c++/8.1.1/bits/stl_tree.h:231:41:   required from ‘struct std::_Rb_tree_node<std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> > >’
/usr/include/c++/8.1.1/bits/stl_tree.h:1872:21:   required from ‘void std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_erase(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Link_type) [with _Key = BasicBlockNode*; _Val = std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> >; _KeyOfValue = std::_Select1st<std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> > >; _Compare = std::less<BasicBlockNode*>; _Alloc = std::allocator<std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> > >; std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Link_type = std::_Rb_tree_node<std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> > >*]’
/usr/include/c++/8.1.1/bits/stl_tree.h:965:9:   required from ‘std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::~_Rb_tree() [with _Key = BasicBlockNode*; _Val = std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> >; _KeyOfValue = std::_Select1st<std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> > >; _Compare = std::less<BasicBlockNode*>; _Alloc = std::allocator<std::pair<BasicBlockNode* const, std::set<boost::shared_ptr<ProgramOperation>, ProgramOperation::Comparator> > >]’
/usr/include/c++/8.1.1/bits/stl_map.h:183:7:   required from here
/usr/include/c++/8.1.1/bits/stl_tree.h:452:21: error: static assertion failed: comparison object must be invocable with two arguments of key type
       static_assert(__is_invocable<_Compare&, const _Key&, const _Key&>{},
```

Seems to be related to [this](https://gcc.gnu.org/gcc-8/porting_to.html):

>  The associative containers (std::map, std::multimap, std::set, and std::multiset) now use static assertions to check that their comparison functions support the necessary operations. In C++17 mode this includes enforcing that the function can be called when const-qualified:

hkultala, 2018-08-15:
What is the TCE source code line where this originates from ? 

CopyingDelaySlotFiller.cc:33 is only #include<set>.

franz, 2018-08-15:
@hkultala that is indeed a mystery... that error is all i get from GCC.

hkultala, 2018-08-22:
The comparator for ProgramOperationPtr is in tce/src/base/program/ProgramOperation.hh:147.

Does changing the parameter types from 

const ProgramOperationPtr po1, const ProgramOperationPtr po2

to references

const ProgramOperationPtr& po1, const ProgramOperationPtr& po2

fix this?

franz, 2018-08-24:
Sorry, it doesn't fix it, but i found out what does. I'll make a PR soon.

commit bdbcb75c (2019-03-18): Fix #73 - compiling with GCC 8.x
closed 2019-03-18 by pjaaskel