cpc/openasip #79: [tcecc] C++11 support?

alekob, 2019-03-21:
Hi, 
I'd like to compile C++11 code on a TTA system, stdc++ lib is necessary. As the installation procedure in TCE for LLVM differs from the "getting started" page (https://llvm.org/docs/GettingStarted.html), I'd like to ask how to install libcxx and libcxxabi to work with tcecc?

pjaaskel, 2019-03-22:
I don't think this has been tried by anyone yet. While I think most c++11 *language features* should just work, no one has ported a C++ standard library so far. Our group has focused on C and OpenCL C. It would be great if someone ported it though!

alekob, 2019-03-26:
Tried to extend the installation script for LLVM 8.0 to compile libc++, libc++abi. The build process seems to work, but the compilation of the project's C++ code results in several errors:

/home/alex/apps/LLVM/8.0/include/c++/v1/__config:63:4: error: Unknown object file format
#  error Unknown object file format
   ^
/home/alex/apps/LLVM/8.0/include/c++/v1/__config:1172:6: error: "No thread API"
#    error "No thread API"
...
 /home/alex/apps/LLVM/8.0/include/c++/v1/atomic:1855:16: error: use of undeclared identifier 'int_least64_t'
typedef atomic<int_least64_t>  atomic_int_least64_t;
 ...

hkultala, 2019-08-05:
TCE does not yet support 64-bit simple data types. This is the reason the int_least64_t is not supported.

alekob, 2019-08-07:
How about the version on branch "64bit"? Does it support simple 64 bit types or is there a way for a "software emulation" like "-swfp"?