cpc/openasip #90: a problem about math.h

VanceVin, 2019-07-08:
this is My test code:
#include <stdio.h>
#include <stdlib.h>
#include <math.h>
   
typedef struct
{
	int width;
	int height;
	int channels;
	unsigned char* imageData;
} ClImage;

int main(int argc, char* argv[])
{

    unsigned char test1[30]={0x60,0x5c,0xaa,0x69,0x65,0xb1,0x69,0x65,0xad,0x6e,0x69,0xae,0x69,0x65,0xa7,0x67,
0x65,0xa1,0x67,0x67,0xa3,0x6d,0x6f,0xab,0x68,0x6c,0xa8,0x65,0x69,0xa5};
            ClImage* img1=(ClImage*)malloc(sizeof(ClImage));
            img1->width=2;
            img1->height=5;
            img1->channels=3;
            img1->imageData = (unsigned char*)malloc(sizeof(unsigned char)*30);
            img1->imageData=test1;

	ClImage* bmpImg = (ClImage*)malloc(sizeof(ClImage));
	bmpImg->imageData = (unsigned char*)malloc(sizeof(unsigned char)*10);
	int i;
	int width=2;
	int height=5;
	int channel=3;
	unsigned char* in_img = img1->imageData;
	for (i = 0; i < width*height; i++)
	{
		float gray = 0.114 * in_img[i*channel] + 0.587 * in_img[i*channel + 1] + 0.299 * in_img[i*channel + 2];
		if (gray - (int)(gray) < 0.5)
			bmpImg->imageData[i] = floorf(gray);
			
		else
			bmpImg->imageData[i] = ceilf(gray);
		printf("imagedata[%d]=%d",i,bmpImg->imageData[i]);
					
	}
return 0;        
}
when i use tcecc , it says :Error: procedure 'ceilf' not found!

pjaaskel, 2019-07-08:
Please add your .adf and the tcecc command line how you try to compile this.

VanceVin, 2019-07-09:
i just use tremor_exercise/adf/three_bus_std_mul.adf, and i have tried it with/without float FU, i have tried start.adf and other adf provided in the exercise.
command line : added or deleted the contents in parentheses，problem remains the same
tcecc (-O3) (--swfp) (-lm)  -a x.adf -o my.tpef my.c

VanceVin, 2019-07-09:
I tried the code below with the same adf and command：tcecc -a x.adf -o my.tpef my.c
#include <stdio.h>
#include <math.h>
int main()
{
float a=1.5258;
int d=floor(a);
float b=round(a);
unsigned char c=ceilf(a);
double f=expf(a);
printf("result is %f\n%d\n%d\n%f\n",b,c,d,f);
return 0;
};

VanceVin, 2019-07-09:
I tried the code below with the same adf and command：tcecc -a x.adf -o my.tpef my.c
#include <stdio.h>
#include <math.h>
int main()
{
float a=1.5258;
int d=floor(a);
float b=round(a);
unsigned char c=ceilf(a);
double f=expf(a);
printf("result is %f\n%d\n%d\n%f\n",b,c,d,f);
return 0;
};
it work well, and the results are exactly true in proxim

pjaaskel, 2019-07-09:
Can you try if the below patch fixes it for you?
```
diff --git a/tce/src/applibs/LLVMBackend/passes/LowerIntrinsics.cc b/tce/src/applibs/LLVMBackend/passes/LowerIntrinsics.cc
index a90322e..b32afd8 100644
--- a/tce/src/applibs/LLVMBackend/passes/LowerIntrinsics.cc
+++ b/tce/src/applibs/LLVMBackend/passes/LowerIntrinsics.cc
@@ -110,6 +110,8 @@ LowerIntrinsics::doInitialization(Module &M) {

     // Initialize list of intrinsics to lower.
     replace_.insert(Intrinsic::flt_rounds);
+    replace_.insert(Intrinsic::ceil);
+    replace_.insert(Intrinsic::floor);
     replace_.insert(Intrinsic::memcpy);
     replace_.insert(Intrinsic::memset);
     replace_.insert(Intrinsic::memmove);
```

VanceVin, 2019-07-12:
// Initialize list of intrinsics to lower.
     replace_.insert(Intrinsic::flt_rounds);
+    replace_.insert(Intrinsic::ceil);
+    replace_.insert(Intrinsic::floor);
+   replace_.insert(Intrinsic::round);
+   replace_.insert(Intrinsic::exp2);
     replace_.insert(Intrinsic::memcpy);
     replace_.insert(Intrinsic::memset);
     replace_.insert(Intrinsic::memmove);

after and these words，and my case works well