cpc/openasip #99: tcecc segfaults in case of an unconnected FU

pjaaskel, 2019-10-16:
I accidentally left the REFLECT FU unconnected in TCE tour. tcecc just segfaults in this case (at least when trying to invoke the custom ops via macros).

Sannoso, 2020-07-13:
What would be the right approach to fix this? What is the wanted behavior?
1) Not being able to save an adf with unconnected FUs
2) Allow adfs with unconnected FUs and when running tcecc, throw an error with a clear message instead of a segfault
3) something else?

pjaaskel, 2020-07-13:
3. just ignore that in RM construction as a resource one can use. But like said, this is not worth looking into at this point when we have more urgent tasks.

Sannoso, 2021-05-29:
see new pull request with fix https://github.com/cpc/tce/pull/122

closed 2022-05-17 by Sannoso