cpc/openasip #103: self test failed

postoroniy, 2020-04-06:
clone latest master at 25c02565e42a075d81366d3411869d6463f38955
followed to install and downloaded and llvm10 by script.
configured, compiled and installed in the same way as per instructions in install.
ran tce-selftest -v and got this
Testing TCE installation, this can take up to two hours.
Time to go for a lunch? Use -v to see the progress.
test_noOpt (__main__.TestCompiler)
Test C compilation with -O0 ... FAIL
test_opt (__main__.TestCompiler)
Test C compilation with -O3 ... FAIL
test_custom_operation_from_c_code (__main__.TestCustomOps)
Test creating a simple custom operation with simulation code and ... ok
test_basic_plugins (__main__.TestExplorer)
Test the ImplementationSelector and Evaluate explorer plugins (this takes a while!) ... ok
test_default_plugins_found (__main__.TestExplorer)
Check that the default Explorer plugins are listed. ... ok
test_proge_and_pig (__main__.TestProGeAndPIG)
Test the processor generator and the program image generator (with dictionary instruction compression) ... FAIL
test_compiled_simulation (__main__.TestSimulator)
Test the compiled engine of the simulator ... FAIL
test_debugging_simulation (__main__.TestSimulator)
Test the debugging engine of the simulator ... FAIL

for all fails same source code mentioned
======================================================================
FAIL: test_noOpt (__main__.TestCompiler)
Test C compilation with -O0
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/slava/local/bin/tce-selftest", line 242, in test_noOpt
    self._test_compilation("-O0 --swfp")
  File "/home/slava/local/bin/tce-selftest", line 235, in _test_compilation
    self.assertEquals(stderr, "")
AssertionError: 'warning: Linking two modules of different data layouts: \'/tmp/tcecc-EAV5tB/tmphNpVVe.o\' is \'e-p:32:32:32-i1:8:8-i8:8:32-i16:16:32-i32:32:32-i64:32:32-f32:32:32-f64:32:32-v64:64:64-v128:128:128-v256:256:256-v512:512:512-v1024:1024:1024-v2048:2048:2048-v4096:4096:4096-a0:0:32-n32\' whereas \'llvm-link\' is \'e-p:32:32:32-i1:8:8-i8:8:32-i16:16:32-i32:32:32-i64:32:32-f32:32:32-f64:32:32-v64:32:32-v128:32:32-v256:32:32-v512:32:32-v1024:32:32-a0:0:32-n32\'\n\nDataLayout mismatch with module: /tmp/tcecc-EAV5tB/tmp0mXs7X.tpef_emul_internalized.bc.intrinsics_lowered\nModule: e-p:32:32:32-i1:8:8-i8:8:32-i16:16:32-i32:32:32-i64:32:32-f32:32:32-f64:32:32-v64:32:32-v128:32:32-v256:32:32-v512:32:32-v1024:32:32-a0:0:32-n32\nTarget: e-p:32:32:32-i1:8:8-i8:8:32-i16:16:32-i32:32:32-i64:32:32-f32:32:32-f64:32:32-v64:64:64-v128:128:128-v256:256:256-v512:512:512-v1024:1024:1024-v2048:2048:2048-v4096:4096:4096-a0:0:32-n32\nllvm-tce: /home/slava/projects/tce/tce/llvm-build-Release/release_10/llvm/lib/CodeGen/MachineFunction.cpp:201: void llvm::MachineFunction::init(): Assertion `Target.isCompatibleDataLayout(getDataLayout()) && "Can\'t create a MachineFunction using a Module with a " "Target-incompatible DataLayout attached\\n"\' failed.\nAborted\n' != ''

pjaaskel, 2020-04-06:
Looks like you might have built TCE against a wrong LLVM, perhaps the one shipped with your Linux distribution, and running it with the TCE built one, or the other way around. You can use 'ldd' too to see which LLVM libtce.so is linked to.

postoroniy, 2020-04-06:
so it means that these lines not quite correct?
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/local/lib
export PATH=$PATH:$HOME/local/bin
as mine prior installed llm/clang would have used to compiled?

pjaaskel, 2020-04-06:
That's a usual problem. You can use 'llvm-config --prefix' to check which LLVM it finds in the path.

postoroniy, 2020-04-07:
thanks, solved now

closed 2020-04-07 by postoroniy