cpc/openasip #113: issue for installing TCE on ubuntu20.04

hitanvil, 2021-01-01:
I installed TCE on Ubuntu20.04 LTS following the guide in http://openasip.org/downloads/INSTALL. But the final self test fails.

# Install LLVM 
    install LLVM by  tools/scripts/install_llvm_11.sh $HOME/local. It reports below information
       dev@TCE-VM:~/tce/tce-1.22/tce$ tools/scripts/install_llvm_11.sh $HOME/local
        ### LLVM build mode: -DCMAKE_BUILD_TYPE=Release -DLLVM_ENABLE_ASSERTIONS=ON
        Already on 'release/11.x'
        Your branch is up to date with 'origin/release/11.x'.
        HEAD is now at 6b458878ac6 Rename llvm-config to llvmtce-config
        Already up to date.
        -- clang project is enabled
        -- clang-tools-extra project is disabled
        -- compiler-rt project is disabled
        -- debuginfo-tests project is disabled
        -- libc project is disabled
        -- libclc project is disabled
        -- libcxx project is disabled
        -- libcxxabi project is disabled
        -- libunwind project is disabled
        -- lld project is disabled
        -- lldb project is disabled
        -- mlir project is disabled
        -- openmp project is disabled
        -- parallel-libs project is disabled
        -- polly project is disabled
        -- pstl project is disabled
        -- flang project is disabled
        -- Native target architecture is X86
        -- Threads enabled.
        -- Doxygen disabled.
        -- Go bindings disabled.
        -- OCaml bindings disabled, need ctypes >=0.4.
        -- LLVM host triple: x86_64-unknown-linux-gnu
        -- LLVM default target triple: x86_64-unknown-linux-gnu
        -- Building with -fPIC
        -- Constructing LLVMBuild project information
        -- Targeting X86
        -- Clang version: 11.0.1
        -- Registering Bye as a pass plugin (static build: OFF)
        -- Failed to find LLVM FileCheck
        -- Version: 0.0.0
        -- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
        -- Performing Test HAVE_POSIX_REGEX -- success
        -- Performing Test HAVE_STEADY_CLOCK -- success
        -- Configuring done
        -- Generating done
        -- Build files have been written to: /home/dev/tce/tce-1.22/tce/llvm-build-Release/release_11/build
        (Below are progress report and seems no problem)
# Check enviroment
         
            dev@TCE-VM:~/tce/tce-1.22/tce$ echo $LD_LIBRARY_PATH
            :/home/dev/local/lib
            dev@TCE-VM:~/tce/tce-1.22/tce$ echo $PATH
            /home/dev/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/media/sf_tools/clion-2020.3/bin
            dev@TCE-VM:~/tce/tce-1.22/tce$ echo $LDFLAGS
            -L/home/dev/local/lib
            dev@TCE-VM:~/tce/tce-1.22/tce$ ./autogen.sh
            Preparing the tce build system...please wait
            
            Found GNU Autoconf version 2.69
            Found GNU Automake version 1.16.1
            Found GNU Libtool version 2.4.6
            
            Automatically preparing build ... done
            
            The tce build system is now prepared.  To build here, run:
              ./configure
              make

# Generate makefile
[... 637 lines cut ...]
            ======================================================================
            FAIL: test_debugging_simulation (__main__.TestSimulator)
            Test the debugging engine of the simulator
            ----------------------------------------------------------------------
            Traceback (most recent call last):
            File "/home/dev/local/bin/tce-selftest", line 260, in test_debugging_simulation
                self._test_simulation("")
            File "/home/dev/local/bin/tce-selftest", line 254, in _test_simulation
                self.assertEqual(stdout.strip(), "Hello \nTTA World!")
            AssertionError: 'Unable to load program. Cannot find suita[60 chars]zed.' != 'Hello \nTTA World!'
            - Unable to load program. Cannot find suitable reader implementation for file.
            - Simulation not initialized.+ Hello 
            + TTA World!

            ----------------------------------------------------------------------
            Ran 8 tests in 18.760s

            FAILED (failures=5)

Seems the LLVM installation has problem. I had tried several times but the result is the same. Are there anything I ingored?

BTW, is it possible to create a TCE docker image or VM disk? It is more easy to skip those system and installation problem.

Regards
Hitanvil

pjaaskel, 2021-01-03:
Thanks for reporting this major issue that was introduced very late in the release process (and was overlooked due to testers 'make install'ing LLVM without removing the old directory with an llvm-config first).  

This should be now fixed in the release_22 branch. Please pull and test and reopen if it's still broken.

For the container or VM, we create a VM with TCE preinstalled occasionally, but simply don't have the time&people to do it for each release regularly.

closed 2021-01-03 by pjaaskel

hitanvil, 2021-01-03:
Thanks a lot, Dr. Jaaskelainen. I will re-install it. It is a really good
tool.

hitanvil, 2021-01-06:
Reinstalled, it works

hitanvil, 2021-01-07:
Hi, Dr. Pekka,

I read the paper TTA-SIMD soft core processors. It is very interesting. I'd
like to know if I can replicate the work in latest TCE environment. The
paper mentioned many previous work such as SIMD, AivoTTA, AXI interface etc
to get this work done. I am not sure if those features already included in
the latest TCE so that I can use those modules/template directly.

I am working on signal processing SW most of time but like to learn some HW
knowledge also. TCE seems to be a good start point. Thanks your great help.

Regards
Xuebin

pjaaskel, 2021-01-07:
Hi Xuebin. The SIMD work described in the FPL is a work in progress and not donated to the open source repository so far. Please send me an email at pekka.jaaskelainen@tuni.fi and describe what type of project you have so we can discuss if there could be a possibility for collaboration regarding this. Thanks!

hitanvil, 2021-01-07:
Thanks a lot, Dr. Pekka. I feel TCE is a good way to learn more HW
as its design methodology more like an SW. I had done many VLIW and SIMD SW
for signal processing but also want to know how to design those algorithms
in HW. If the SIMD is not open now, I will use current TCE to learn its
methodology.

Regards
Xuebin

commit ddaedfe9 (2021-01-03): Use correct llvm{,tce]-config.

This bug basically broke usage with fresh LLVMTCE 11 installs
since it assumed llvm-config is available although it was
renamed to llvmtce-config in TCE-specific LLVM repo created
late in the 1.22 release cycle.

Fixes #113.