cpc/openasip #136: tce-selftest fails on Python >= 3.10 due to deprecation warning

sarahec, 2022-05-10:
Beginning with Python 3.10, Python prints a deprecation warning for any use of distutils. Since the selftest checks for an empty stdout to pass the test, this triggers a failure.

 `tcecc` appears to be the culprit.

The fix is to switch to `setuptools`.

FAIL: test_noOpt (__main__.TestCompiler)
Test C compilation with -O0
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/sec/local/bin/tce-selftest", line 199, in test_noOpt
    self._test_compilation("-O0 --swfp")
  File "/home/sec/local/bin/tce-selftest", line 192, in _test_compilation
    self.assertEqual(stderr, "")
AssertionError: '/home/sec/local/bin/tcecc:40: Deprecation[187 chars]le\n' != ''
- /home/sec/local/bin/tcecc:40: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
-   from distutils.spawn import find_executable

sarahec, 2022-05-10:
I found the problem, and it's a trivial fix. Line 40 of `/src/bintools/Compiler/tcecc.in` reads:
```python
from distutils.spawn import find_executable
```
but the code was fixed to no longer use distutils. Removing this line and rebuilding solves the problem.

It's probably quicker for you to make the edit than merge a PR, but let me know if you want the PR.

pjaaskel, 2022-05-11:
Thanks @sarahec -- a PR would be great so you get all the glory in 'git blame' etc.! :)

sarahec, 2022-05-12:
Whoops, merged to my personal copy! Let me send this in properly.

closed 2022-05-13 by sarahec