cpc/openasip #141: UX question: do the numeric spinners make sense?

sarahec, 2022-05-15:
I'm noticing that the numeric inputs have spinners (-|+) on them in places where you're looking for an arbitrary numeric value. They're not particularly useful and are taking up space. 

Would it make more sense to attach a numeric [validator](https://docs.wxwidgets.org/3.0/overview_validator.html) to each of these and scrap the spinner?

(Yes, I'm volunteering)

pjaaskel, 2022-05-16:
Yes! The UX of the GUIs is terrible in some places - all contributions to improve the situation very much welcomed.

sarahec, 2022-05-17:
Here's my execution plan. Starting with ProDe as the initial case:

1. Clean up GUI assertion errors (which should have no effect on the UI except allow it to be used on other distros)
2. Fix any crashes (e.g. the broken imagelist)
3. Remove the width limitation on wxSpinControl as it's not scaling the control as may have been expected.
4. Get your review on the above
5. Remove the spin controls, adding numeric validation to those text fields. (Also better for accessibility)
6. Clean up position & alignments within dialogs to make them a bit more polished. (Not changing the workflow).

Then I can extend this work through the codebase and we can consider next steps.

pjaaskel, 2022-05-18:
Sounds great!

sarahec, 2022-07-01:
Replacing the spinners with validated numeric inputs isn't a straightforward conversion after all. Since they already work with the keyboard, I'm going to close this.

closed 2022-07-01 by sarahec