cpc/openasip #175: Assertion `!instructionStartCache_.empty()' failed

nrother, 2022-08-18:
When trying to compile a program using tcecc I get the following error message:

```
rother@pikachu:~/Documents/TTA/GBP_TTA/tta_bug$ tcecc --version
tcecc - TCE retargetable compiler 1.25
rother@pikachu:~/Documents/TTA/GBP_TTA/tta_bug$ tcecc -x c++ -O3 -a baseline.adf -o gbp.tpef gbp.cpp
llvm-tce: CodeSection.cc:131: void TPEF::CodeSection::initInstructionCache() const: Assertion `!instructionStartCache_.empty()' failed.
```

This looks like a compiler bug to me.

Minimal example to reproduce this bug:

gbp.cpp
```cpp
#include "fpm_tta/fixed.hpp"
namespace ff = fpm;
using position_t = fpm::fixed<int32_t, int32_t, 16>;

volatile int32_t antennaPositions[20];

int main()
{
    position_t *antennaPositions = (position_t*)antennaPositions;
    
    position_t beamlimitMaxRangeDist = antennaPositions[1];

    volatile int *tmp = (volatile int*)0x200;
    *tmp = (volatile int)beamlimitMaxRangeDist;

    return 0;
}
```

fpm_tta/fixed.hpp
```cpp
#ifndef FPM_FIXED_HPP
#define FPM_FIXED_HPP

#include <stdint.h> //note: not int64_t on TTA!

namespace fpm
{

template <typename BaseType, typename IntermediateType, unsigned int FractionBits>
class fixed
{
    static constexpr BaseType FRACTION_MULT = BaseType(1) << FractionBits;

public:
    template <typename T>
    constexpr inline explicit operator T() const noexcept
    {
        return static_cast<T>(m_value) / FRACTION_MULT;
    }

private:
    BaseType m_value;
};

} // namespace fpm

#endif
```

Let me know if you need any further information!

nrother, 2022-08-18:
Interestingly, it fails in version 1.22 with a different error message:

```
rother@pikachu:~/Documents/TTA/GBP_TTA/tta_bug$ tcecc --version
tcecc - TCE retargetable compiler 1.22
rother@pikachu:~/Documents/TTA/GBP_TTA/tta_bug$ tcecc -x c++ -O3 -a baseline.adf -o gbp.tpef gbp.cpp
Error: procedure '_exit' not found!
```

I have no tested it with the latest version from git.

pjaaskel, 2022-08-18:
> Let me know if you need any further information!

Just a pull request to fix it will do :) Just kidding. The C++ support is not much tested and far from stable, so either you have to be prepared to stabilize the code related to that and contribute back to OpenASIP or use C.