cpc/openasip #183: Question about tooling

robertkeizer, 2022-10-27:
Heyo - quick question; I'm experimenting with `openasip` ( thanks ), and have got a rudimentary docker image going that contains the toolchain passing tests ( https://github.com/reconfigurable/openasip ) .

It's really rough right now, and I'm exposing the graphical interface over `x11vnc`.

Is this something I should clean up and create a PR back into `openasip` proper?

Basically asking if there would be any appetite for a `Dockerfile` and related scripts being included in the project itself.

pjaaskel, 2022-10-27:
Yes, it would be definitely helpful to other users. Thanks in advance!

robertkeizer, 2022-11-10:
Any back and forth around the implementation of this can be done over in #190 - closing this issue.

closed 2022-11-10 by robertkeizer