cpc/openasip #200: Mismatches between xilinx_basic_alu.vhd and generic_sru.vhd

ISDIES, 2022-11-18:
Hi to all! I'm pretty new at Github, so I hope this is the correct way to raise my problem.
I'm trying to create a new TTA processor which features an alu implemented using xilinx_basic_alu.vhd and generic_sru.vhd (is already present in xilinx_series7.hdb). The problem is that, when it comes the time to compile the HDL (using for example Vivado), the compiler tells me that some mismatch are present between xilinx_basic_alu.vhd and generic_sru.vhd. Fixing them manually, the system is working. To solve this issue, I tried to change the source code which I have under tce-level/tce/hdb/xilinx_vhdl and generate again the processor, but mismatches still occurs. Is there a way to fix the problem? Am I missing something? Thanks

pjaaskel, 2022-11-21:
@TopiLeppanen do you know what's this about?

TopiLeppanen, 2022-11-21:
It seems that the xilinx_basic_alu.vhd was out-dated. Some changes made to the generic_sru.vhd component weren't propagated to xilinx_basic_alu.vhd correctly. Fixing the VHDL of xilinx_basic_alu.vhd seems to fix the problem for me, and now I can generate processors that compile at least. I haven't used the xilinx_basic_alu.vhd component myself, so I can only hope that it works. (Nobody else had probably either, since it's been broken like this years).

If I had to guess why you changing the source codes didn't work, maybe you didn't run 'make install?. You can run it directly in the hdb-folder to only install the updated hdbs and sources. The generateprocessor will then pick up the hdbs and VHDL source files from the install directory.

ISDIES, 2022-11-21:
Thanks @TopiLeppanen ! I was not running the 'make install' command to update my fixes on the VHDL.

PR #202 (2022-11-23): Fix xilinx_basic_alu's interface to generic_sru
Fixes #200 

Can @ISDIES confirm?
closed 2022-11-23 by karihepola