cpc/openasip #211: New version 2.0 oa-selftest -v test error

dingzex, 2022-11-29:
安装后运行 oa-selftest -v 时，发生如下错误：
Traceback (most recent call last):
  File "/home/carbon/local/bin/oa-selftest", line 597, in test_proge_and_pig
    self.assertEqual(stderr, '')
AssertionError: "HDBManager.cc:4391: In function 'int HDB[241 chars]d)\n" != ''
- HDBManager.cc:4391: In function 'int HDB::HDBManager::addBooleanColumn(const string&, const string&)': DEBUG: SQL logic error
- generateprocessor: HDBManager.cc:4392: int HDB::HDBManager::addBooleanColumn(const string&, const string&): Assertion `false' failed.
- Aborted (core dumped)

----------------------------------------------------------------------
Ran 8 tests in 133.643s

FAILED (failures=1)
/home/carbon/图片/2022-11-29 23-49-06 的屏幕截图.png

dingzex, 2022-11-29:
sqlite3 -version is :
3.31.1 2020-01-27 19:55:54 3bfa9cc97da10598521b342961df8f5f68c7388fa117345eeb516eaa837balt1

pjaaskel, 2022-11-30:
Hi, which Linux distribution you are using?

dingzex, 2022-11-30:
> Hi, which Linux distribution you are using?

ubuntu 20.04

pjaaskel, 2022-11-30:
Same here, and it works, hmm.

bojle, 2023-03-15:
Any updates on this issue @pjaaskel ?
I am facing this very error, ubuntu version `22.10`, sqlite version `3.39.3`

bojle, 2023-03-16:
Hey,
what do you mean by latest revision?
I've tried both, llvm 15 (from main branch) and 16 builds (from llvm_16 branch). Both builds suffer from the same issue (SQL logic error in HDBManager)
Here's what i got from `oa-selftest -v` on llvm_16 based build:
```
Testing OpenASIP installation, this can take up to two hours.
Time to go for a lunch? Use -v to see the progress.
test_noOpt (main.TestCompiler)
Test C compilation with -O0 ... FAIL
test_opt (main.TestCompiler)
Test C compilation with -O3 ... ok
test_custom_operation_from_c_code (main.TestCustomOps)
Test creating a simple custom operation with simulation code and ... ok
test_basic_plugins (main.TestExplorer)
Test the ImplementationSelector and Evaluate explorer plugins (this ... ok
test_default_plugins_found (main.TestExplorer)
Check that the default Explorer plugins are listed. ... ok
test_proge_and_pig (main.TestProGeAndPIG)
Test the processor generator and the program image generator (with ... FAIL
test_compiled_simulation (main.TestSimulator)
Test the compiled engine of the simulator ... FAIL
test_debugging_simulation (main.TestSimulator)
Test the debugging engine of the simulator ... FAIL

======================================================================
FAIL: test_noOpt (main.TestCompiler)
Test C compilation with -O0

Traceback (most recent call last):
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 200, in test_noOpt
self._test_compilation("-O0 --swfp")
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 193, in _test_compilation
self.assertEqual(stderr, "")
AssertionError: 'Traceback (most recent call last):\n Fil[355 chars]ed\n' != ''

Traceback (most recent call last):
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oacc", line 1700, in <module>
emulationLinkedFileName = linkEmulationFuncs(optimizedFileName, namePrefix)
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oacc", line 780, in linkEmulationFuncs
" -f " + internalizeCmd + " -o " + outputName +
NameError: name 'internalizeCmd' is not defined
======================================================================
FAIL: test_proge_and_pig (main.TestProGeAndPIG)
Test the processor generator and the program image generator (with

Traceback (most recent call last):
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 597, in test_proge_and_pig
self.assertEqual(stderr, '')
AssertionError: "[HDBManager.cc:4385](http://hdbmanager.cc:4385/): In function 'int HDB[261 chars]d)\n" != ''

[HDBManager.cc:4385](http://hdbmanager.cc:4385/): In function 'int HDB::HDBManager::addBooleanColumn(const std::string&, const std::string&)': DEBUG: SQL logic error
generateprocessor: [HDBManager.cc:4387](http://hdbmanager.cc:4387/): int HDB::HDBManager::addBooleanColumn(const std::string&, const std::string&): Assertion `false' failed.
Aborted (core dumped)
======================================================================
FAIL: test_compiled_simulation (main.TestSimulator)
Test the compiled engine of the simulator

Traceback (most recent call last):
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 265, in test_compiled_simulation
self._test_simulation("-q")
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 255, in _test_simulation
self.assertEqual(stdout.strip(), "Hello \nTTA World!")
AssertionError: 'Unable to load program. Cannot find suita[60 chars]zed.' != 'Hello \nTTA World!'

Unable to load program. Cannot find suitable reader implementation for file.
Simulation not initialized.+ Hello
TTA World!
======================================================================
FAIL: test_debugging_simulation (main.TestSimulator)
Test the debugging engine of the simulator

Traceback (most recent call last):
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 261, in test_debugging_simulation
self._test_simulation("")
File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 255, in _test_simulation
self.assertEqual(stdout.strip(), "Hello \nTTA World!")
AssertionError: 'Unable to load program. Cannot find suita[60 chars]zed.' != 'Hello \nTTA World!'

Unable to load program. Cannot find suitable reader implementation for file.
Simulation not initialized.+ Hello
TTA World!
Ran 8 tests in 22.101s

FAILED (failures=4)
```
I dug a little deep and found the values of `table` and `newcolum` passed to `addBooleanColumn` when it crashes. `table = rf_architecture` and `column = zero_register`. Could the problem be with this table not being present as it is a "logic error"?

pjaaskel, 2023-03-16:
I mean did you try OpenASIP main branch or just the release branch? @nrother this sounds it could be related to your DB fixes.

nrother, 2023-03-17:
Hm, could be, but #213 doesn't touch anything called `rf_architecture` or `zero_register`... Is there a way to get more output from SQlite what the actual problem is?

bojle, 2023-03-17:
Well, I seem to have fixed it mostly. Not really sure how, I did downgrad sqlite to 3.31.1 and do a clean re-compile. 
`oa-selftest` still isn't clean. I'll post the output when I get back to my computer.

pjaaskel, 2023-03-17:
Is this with the master branch or the 2.0 release branch?

bojle, 2023-03-20:
This is with the llvm_16 branch, but compiled with `install_llvm_15.sh`. 
Here's the `oa-selftest`
```
Testing OpenASIP installation, this can take up to two hours.
Time to go for a lunch? Use -v to see the progress.
test_noOpt (__main__.TestCompiler)
Test C compilation with -O0 ... FAIL
test_opt (__main__.TestCompiler)
Test C compilation with -O3 ... ok
test_custom_operation_from_c_code (__main__.TestCustomOps)
Test creating a simple custom operation with simulation code and ... ok
test_basic_plugins (__main__.TestExplorer)
Test the ImplementationSelector and Evaluate explorer plugins (this ... ok
test_default_plugins_found (__main__.TestExplorer)
Check that the default Explorer plugins are listed. ... ok
test_proge_and_pig (__main__.TestProGeAndPIG)
Test the processor generator and the program image generator (with ... FAIL
test_compiled_simulation (__main__.TestSimulator)
Test the compiled engine of the simulator ... FAIL
test_debugging_simulation (__main__.TestSimulator)
Test the debugging engine of the simulator ... FAIL

======================================================================
FAIL: test_noOpt (__main__.TestCompiler)
Test C compilation with -O0
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 200, in test_noOpt
    self._test_compilation("-O0 --swfp")
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 193, in _test_compilation
    self.assertEqual(stderr, "")
AssertionError: 'Traceback (most recent call last):\n  Fil[355 chars]ed\n' != ''
- Traceback (most recent call last):
-   File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oacc", line 1700, in <module>
-     emulationLinkedFileName = linkEmulationFuncs(optimizedFileName, namePrefix)
-   File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oacc", line 780, in linkEmulationFuncs
-     " -f " + internalizeCmd + " -o " + outputName +
- NameError: name 'internalizeCmd' is not defined

======================================================================
FAIL: test_proge_and_pig (__main__.TestProGeAndPIG)
Test the processor generator and the program image generator (with
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 597, in test_proge_and_pig
    self.assertEqual(stderr, '')
AssertionError: "Unable to copy file /tmp/tmp5q9xoier/pro[168 chars]l'\n" != ''
- Unable to copy file /tmp/tmp5q9xoier/proc_vhdl/vhdl/lsu_le.vhdl:Unable to copy '/home/vicharak/hdd/openasip/openasip/llvm/share/openasip/hdb/vhdl/fu/lsu_le.vhdl' to '/tmp/tmp5q9xoier/proc_vhdl/vhdl/lsu_le.vhdl'

======================================================================
FAIL: test_compiled_simulation (__main__.TestSimulator)
Test the compiled engine of the simulator
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 265, in test_compiled_simulation
    self._test_simulation("-q")
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 255, in _test_simulation
    self.assertEqual(stdout.strip(), "Hello \nTTA World!")
AssertionError: 'Unable to load program. Cannot find suita[60 chars]zed.' != 'Hello \nTTA World!'
- Unable to load program. Cannot find suitable reader implementation for file.
- Simulation not initialized.+ Hello
+ TTA World!

======================================================================
FAIL: test_debugging_simulation (__main__.TestSimulator)
Test the debugging engine of the simulator
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 261, in test_debugging_simulation
    self._test_simulation("")
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 255, in _test_simulation
    self.assertEqual(stdout.strip(), "Hello \nTTA World!")
AssertionError: 'Unable to load program. Cannot find suita[60 chars]zed.' != 'Hello \nTTA World!'
- Unable to load program. Cannot find suitable reader implementation for file.
- Simulation not initialized.+ Hello
+ TTA World!

----------------------------------------------------------------------
Ran 8 tests in 25.471s

FAILED (failures=4)
```

pjaaskel, 2023-03-20:
Please retry with the 'main' branch. The llvm_16 branch is a WiP branch which might break.

bojle, 2023-03-20:
Built with 'main' branch, install_llvm_15.sh, autogen -> configure -> make -> make install. 
```
$ oa-selftest -v
Testing OpenASIP installation, this can take up to two hours.
Time to go for a lunch? Use -v to see the progress.
test_noOpt (__main__.TestCompiler)
Test C compilation with -O0 ... ok
test_opt (__main__.TestCompiler)
Test C compilation with -O3 ... ok
test_custom_operation_from_c_code (__main__.TestCustomOps)
Test creating a simple custom operation with simulation code and ... ok
test_basic_plugins (__main__.TestExplorer)
Test the ImplementationSelector and Evaluate explorer plugins (this ... ok
test_default_plugins_found (__main__.TestExplorer)
Check that the default Explorer plugins are listed. ... ok
test_proge_and_pig (__main__.TestProGeAndPIG)
Test the processor generator and the program image generator (with ... FAIL
test_compiled_simulation (__main__.TestSimulator)
Test the compiled engine of the simulator ... ok
test_debugging_simulation (__main__.TestSimulator)
Test the debugging engine of the simulator ... ok

======================================================================
FAIL: test_proge_and_pig (__main__.TestProGeAndPIG)
Test the processor generator and the program image generator (with
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/vicharak/hdd/openasip/openasip/llvm/bin/oa-selftest", line 597, in test_proge_and_pig
    self.assertEqual(stderr, '')
AssertionError: "HDBManager.cc:4377: In function 'int HDB[261 chars]d)\n" != ''
- HDBManager.cc:4377: In function 'int HDB::HDBManager::addBooleanColumn(const std::string&, const std::string&)': DEBUG: SQL logic error
- generateprocessor: HDBManager.cc:4378: int HDB::HDBManager::addBooleanColumn(const std::string&, const std::string&): Assertion `false' failed.
- Aborted (core dumped)

----------------------------------------------------------------------
Ran 8 tests in 49.963s

FAILED (failures=1)
```

pjaaskel, 2023-03-20:
@karihepola that HDB zero_registers change was done with the RISC-V features. Do you have ideas?

@bojle is it possible that you have installed an older version of the tools and it somehow finds the older one's HDB (without that column) somewhere?

karihepola, 2023-03-20:
I tried triggering this issue myself but was not successful. OpenASIP should be able to migrate older versions of the HDBs without that zero_register column to the newer version wher that column is present. (I reckon that is why it was called in this case). However, the repository should have updated all the included HDBs to the newest version with that column added.

@bojle Is it possible that you have somehow corrupted HDBs? I recommend you try cloning the repository again if you haven't yet.

karihepola, 2023-03-20:
Then your HDBs are most likely fine. You can find the included HDBs (*.hdb files) in the openasip/openasip/hdb directory. Were you able to find out which hdb library causes the crash? All of them should have that rf_architecture table included.

karihepola, 2023-03-21:
For some reason it is not finding that lsu_le.vhdl source file in your install directory. If you manually check can you see it there in the mentioned path? You can find the source file [here](https://github.com/cpc/openasip/blob/main/openasip/hdb/vhdl/fu/lsu_le.vhdl) and it should be moved to your install directory during the installation phase.