cpc/openasip #222: where to get the proxim and prode tools?

xtfan, 2022-12-20:
Hi, we have sucessfully installed the openasip  according to the README, and try to run the demo step by step according to the tutorial.  However, I can not find the simulator proxim and the architecture viewer prode. Where to get them?

pjaaskel, 2022-12-20:
Hi, They are built and installed by default to the same bin where the rest of the tools, unless you miss the wxWidgets requirements. It should say so in that case at the end of ./configure. Perhaps run the ./configure again, check if it says that and install the missing wxWdigets library requirements as suggested in the README.md?

xtfan, 2022-12-21:
Hi, @pjaaskel. After install wxWidgets correctly,  I can get the tool proxim and prode to continue the tutorial. Thanks!

pjaaskel, 2022-12-21:
Good to hear!

closed 2022-12-21 by pjaaskel