cpc/openasip #230: Installation problem

Remaxic, 2023-04-14:
Hi,
I'm trying to install OpenASIP.But while i'm running this command  ./autogen.sh && ./configure --prefix=$HOME/local && make -j8 && make install  , the terminal window keeps flashing back before the end of the run. 
I don't know how to solve it and am very eager for your help. Thanks and wish you a good day!

pjaaskel, 2023-04-14:
Hi, in order for us to give you tips, you need to provide more info than "flashing back", that is, what was the error message etc.

Remaxic, 2023-04-14:
Thank you very  much for your reply！！ I really want to provide the error msg but actually the terminal window and sometimes even the virtual machine(VMware Ubuntu 22.04)  just shut down suddenly. My screen recording shows that the last process before the terminal exited abnormally is the CopyingDelayslotFiller.o 's compile and some blue "notes" were printed.
During the running I saw many "notes" and "warnings", but I think this is not the cause of the sudden abnormal exit of the terminal window.

pjaaskel, 2023-04-14:
OK. It sounds your VM ran out of memory. Try reducing the parallel builds (run only 'make', not 'make -j8').

Remaxic, 2023-04-17:
Thank you very much！Using your method，my terminal has never exited abnormally again. But now there is a new problem and the error msg is ‘OperationPoolPimpl.cc:60:10: fatal error: llvm/MC/MCInstrDesc.h: No such file or directory’.

Remaxic, 2023-04-17:
Thank you very much! My terminal has never exited abnormally again.But now I encounter a new error like 'OperationPoolPimpl.cc:60:10: fatal error: llvm/MC/MCInstrDesc.h: No such file or directory'.
I installed llvm using the installation instructions mentioned in the README f and there were no errors during the installation. I also added environment variables and updated them with 'source'. But when I tried to run 'clang' to test whether llvm was installed successfully, the terminal told me that it could not find the command.

Remaxic, 2023-04-18:
I 've solved this problem! Thank you again!

closed 2023-04-18 by pjaaskel