cpc/openasip #236: Error compiling risc-v source code

claubersouza, 2023-07-28:
I'm following the manual to learn how to better use the tool for Risc-V. 

$oacc-riscv -O3 -a start.adf --output-format=bin -o crc.img main.c crc.c

ld.lld: error: unable to find library -lc
ld.lld: error: unable to find library -lm

Mode Debug:
$ oacc-riscv -D_DEBUG -O3 -a start.adf --output-format=bin -o crc.img main.c crc.c

main.c:23:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.

karihepola, 2023-08-01:
Did you install all the prerequisites described in [README.MD](https://github.com/cpc/openasip/blob/main/README.md) before compiling OpenASIP? It looks like newlib was not compiled for RISC-V. I recommend making sure RISC-V GNU Toolchain is installed and in PATH then doing `make distclean` or cloning a clean repository before compiling and installing OpenASIP again.

claubersouza, 2023-08-25:
I installed Open Asip again and the following error is occurring

oacc-riscv -D_DEBUG -O3 -a start.adf --output-format=bin -o crc.img main.c crc.c
main.c:49:55: warning: format specifies type 'unsigned int' but the argument has type 'crc' (aka 'unsigned long') [-Wformat]
        iprintf("The crcFast() of \"TCE rocks!\" is 0x%X\n", result);
                                                      ~~     ^~~~~~
                                                      %lX
1 warning generated.

ld.lld -lc -lm -lgcc    -L /opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0 -L /home/clauber/local/riscv/lib  -Ttext=0 -T /tmp/oacc-x78hjiw8/riscv.ld /tmp/oacc-x78hjiw8/init.o /tmp/oacc-x78hjiw8/tmpfrtbean_.o -o /tmp/oacc-x78hjiw8/tmppip_n5ae.hex
ld.lld: error: unable to find library -lc
ld.lld: error: unable to find library -lm

karihepola, 2023-08-25:
It still looks like the same issue. The standard libraries should get installed to `/home/clauber/local/riscv/lib` when OpenASIP is installed with RISC-V support. It is required, however, that all the RISC-V prerequisites are installed and in PATH before configuring/compiling OpenASIP. Can you show the output of `./configure` ?

claubersouza, 2023-08-26:
openasip-2.0/openasip$ ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether make sets $(MAKE)... (cached) yes
checking for a sed that does not truncate output... /usr/bin/sed
checking how to print strings... printf
checking for a sed that does not truncate output... (cached) /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
[... 53 lines cut ...]

riscv-gnu-toolchain$ ./configure
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed

karihepola, 2023-08-28:
The ./configure output seems ok. Can you go to `newlib-1.17.0/riscv-llvm` and call `make && make install` manually after installing the toolset and see if this fixes the issue?

claubersouza, 2023-08-29:
~/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm$ make
make[1]: Entering directory '/home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm'
make[1]: Nothing to be done for 'all-host'.
Checking multilib configuration for newlib...
make[2]: Entering directory '/home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm/riscv/newlib'
/bin/bash ./config.status --recheck
running /bin/bash /home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/newlib/configure  --cache-file=./config.cache --with-newlib --with-cross-host=x86_64-unknown-linux-gnu --prefix=/usr/local --disable-multilib --disable-libgloss --program-transform-name=s,^,riscv-, --with-target-subdir=riscv --build=x86_64-unknown-linux-gnu --host=riscv --target=riscv --srcdir=/home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/newlib CC=/home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm/../../src/bintools/Compiler/oacc-riscv -B/home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm/riscv/newlib/ -isystem /home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm/riscv/newlib/targ-include -isystem /home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/newlib/libc/include CFLAGS=--tce-build-mode -D__riscv_float_abi_soft -D__riscv_xlen=32 -O2 -fno-builtin --disable-internalize -Wno-unknown-warning-option -Wno-literal-range -Wno-pointer-sign -Wno-return-type -Wno-empty-body -Wno-dangling-else -Wno-implicit-function-declaration -Wno-unused-value   CPPFLAGS= LDFLAGS= build_alias=x86_64-unknown-linux-gnu host_alias=riscv target_alias=riscv  --no-create --no-recursion
configure: loading cache ./config.cache
checking build system type... (cached) x86_64-unknown-linux-gnu
checking host system type... (cached) riscv-none
configure: error: invalid value of canonical host
make[2]: *** [Makefile:444: config.status] Error 1
make[2]: Leaving directory '/home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm/riscv/newlib'
make[1]: *** [Makefile:8785: all-target-newlib] Error 2
make[1]: Leaving directory '/home/clauber/dev/fei/openasip-2.0/openasip/newlib-1.17.0/riscv-llvm'
make: *** [Makefile:729: all] Error 2

claubersouza, 2023-08-29:
ok, I will reinstall Opensip again, following the steps described in the README. Thank you very much for your help, I hope to bring good news

claubersouza, 2023-08-29:
Do you have any virtual machine image with OpenASIP installed? @karihepola

claubersouza, 2023-09-01:
I installed it but I still get the error: 

(base) clauber@clauber-hp250g8notebookpc:~/dev/fei/openasip/tce_tutorials/tce_tour$ oacc-riscv -O3  -a start.adf --output-format=bin -o crc.img main.c crc.c 
ld.lld: error: unable to find library -lc
ld.lld: error: unable to find library -lm

claubersouza, 2023-09-12:
Hello, after a lot of work I managed to install and do a RISC-V tutorial. I used Ubuntu 20.04 and used VirtualBox, with just 1 CORE. Because I saw that the risc-v installation script had Make -j (core number) and this was causing it to not compile correctly. Just Doing Make Worked. Thanks @karihepola

karihepola, 2023-09-12:
Hi! Time went by, sorry for not more actively helping. I am glad you got it to work! It sounds interesting that the GNU toolchain would fail a parallel build. Did it emit some error with a parallel build or did it finish successfully?

claubersouza, 2023-09-12:
finish successfully

commit d5a61a0e (2023-09-13): Disable parallel build in RISC-V tools script by default. Fixes #236
closed 2023-09-13 by karihepola