cpc/openasip #237: Simulation error in RISCV tutorial

TerryGuo, 2023-08-11:
Hi there,
I am following the steps in section 3.14.3 of OpenASIP manual on Ubuntu 22.04 and ended up with error like below, 
```
terry@HPStar:~/projects/tce_tutorials/riscv-proc$ ./ghdl_simulate.sh -r 100000
tb/synch_sram.vhdl:187:11:@0ms:(assertion note): Memory initialization succesful
tb/synch_byte_mask_sram.vhdl:216:11:@0ms:(assertion note): Memory initialization succesful
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@10ns:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
gcu_ic/rv32_microcode_wrapper.vhdl:547:5:@2930ns:(assertion failure): Simulation Finished!
/usr/bin/ghdl-mcode:error: assertion failed
in process .testbench(behavior).dut@proc(structural).tta_core@tta0(structural).rv32_microcode_wrapper_i@rv32_microcode_wrapper(rtl).detect_simulation_end
  from: process work.rv32_microcode_wrapper(rtl).detect_simulation_end at rv32_microcode_wrapper.vhdl:547
/usr/bin/ghdl-mcode:error: simulation failed

``` 

I am new to this OpenASIP tool. Any help are welcomed.

karihepola, 2023-08-11:
Hi,

It looks like the simulation was run successfully. An assertion is used in the RTL to detect when the program has been executed. It is handled as an error, which might look confusing. The simulator prints `gcu_ic/rv32_microcode_wrapper.vhdl:547:5:@2930ns:(assertion failure): Simulation Finished!`, which means that the simulation was run to completion.

TerryGuo, 2023-08-12:
@karihepola I see now. Thanks for help.

closed 2023-08-12 by TerryGuo