cpc/openasip #245: 3.3 Streaming I/O code in the manual- iprintf and while (1) loop issues

Rhrifateee, 2023-10-05:
[Streaming_issues.docx](https://github.com/cpc/openasip/files/12821558/Streaming_issues.docx)
I copied the stdout.adf and added stream_in_stream_in_status stream_out_stream_out_status. Both stream_in and stream_out are important to print out the mathematical operation without using iprintf so that an efficient processor with a smaller number of cycles can be made.

But I am getting unusual results in stream_out and TTA simulator when I am doing any mathematical operation (add, subtract, mult, division) inside the “while (1)” loop.

Also, when I am using ‘iprintf’ to print the results with a for loop, the processor behaves differently and prints unusual digits.

I don’t want to design a processor that can print. But without iprintf, tcecc optimizes the code.

I have added some screenshots in the doc file.

karihepola, 2023-10-06:
Could you provide the adf and source code you are using to better try to reproduce the issue?

Rhrifateee, 2023-10-06:
> Could you provide the adf and source code you are using to better try to reproduce the issue?
I am sharing a zip file that includes source code, adf, and tpef file.
Please let me know if you can open the files or not.
[stream_example.zip](https://github.com/cpc/openasip/files/12832299/stream_example.zip)

Rhrifateee, 2023-10-06:
> Could you provide the adf and source code you are using to better try to reproduce the issue?

This the actual source code from 3.3 Streaming I/O section of the manual.
I modified the code in the given source file of the zip folder.

int main()
{
char byte;
int status;
while (1)
{
_OA_STREAM_IN_STATUS(0, status);
if (status == 0)
break;
_OA_STREAM_IN(0, byte);
_OA_STREAM_OUT(byte);
}
return 0;
}

karihepola, 2023-10-07:
I checked the docx examples. Here's some comments:

Case 1:
    The behavior seems as expected. The "\n" directive adds a new line after each char.

Cases 2-4:
    At a quick glance, these seem to have the same problem. You are reading char by char and adding values to them. Some of these chars are '\n" which causes weird behavior when you are doing arithmetics on them. Note that the stream_in operation indeed reads a char from the input file, not the whole line. Also, you should initialize the iterator i, at least in case 2.

Rhrifateee, 2023-10-07:
> I checked the docx examples. Here's some comments:
>
> Case 1: The behavior seems as expected. The "\n" directive adds a new line after each char.
>
> Cases 2-4: At a quick glance, these seem to have the [...]

I used an iterator, the '\n' line and did this simple arithmetic "samples[i]=byte+b;" inside for loop. The stream out output is fine, but what's wrong with the output in TTA simulator? It's not printing the output properly.

#include<tceops.h>
#include<stdio.h>
int main()
{
char byte;
int status;
int i;
char b=2;
char samples[10];
while (1)
{
_OA_STREAM_IN_STATUS(0, status);
if (status == 0)
break;
_OA_STREAM_IN(0, byte);
_OA_STREAM_OUT(byte);
for (i=0; i<10; i++)
{
samples[i]=byte+b;
iprintf("%c\n", samples[i]);
}
}
return 0;
}
[image]

Rhrifateee, 2023-10-07:
> I checked the docx examples. Here's some comments:
>
> Case 1: The behavior seems as expected. The "\n" directive adds a new line after each char.
>
> Cases 2-4: At a quick glance, these seem to have the [...]

> I checked the docx examples. Here's some comments:
>
> Case 1: The behavior seems as expected. The "\n" directive adds a new line after each char.
>
> Cases 2-4: At a quick glance, these seem to have the [...]

> I checked the docx examples. Here's some comments:
>
> Case 1: The behavior seems as expected. The "\n" directive adds a new line after each char.
>
> Cases 2-4: At a quick glance, these seem to have the [...]
The adf file is shared here:
[stream_example.zip](https://github.com/cpc/openasip/files/12838250/stream_example.zip)

With the following code, the stream in data is properly copied in the stream out:
int main()
{
char byte;
int status;
while (1)
{
_OA_STREAM_IN_STATUS(0, status);
if (status == 0)
break;
_OA_STREAM_IN(0, byte);
_OA_STREAM_OUT(byte);
}
return 0;
}
[image]
[image]

But in the following code, with addition operation, the stream out provides random output:
int main()
{
char byte;
int status;
char b=2;
while (1)
{
_OA_STREAM_IN_STATUS(0, status);
if (status == 0)
break;
_OA_STREAM_IN(0, byte);
byte=byte+b;
_OA_STREAM_OUT(byte);
}
return 0;
}
[image]
[image]

Why does it happen the arithmetic operation is done inside while (1) loop, the stream out provides random output?

karihepola, 2023-10-07:
> But in the following code, with addition operation, the stream out provides random output: int main() { char byte; int status; char b=2; while (1) { _OA_STREAM_IN_STATUS(0, status); if (status == 0) [...]

It's because of the new line directive between entries. If your input file has just one line with `123`, the simulator will output `345` because it reads the input stream char by char and in that case it would sequentially read '1', '2' and '3' and in your example code add two to each read char that is written in the output file.

Rhrifateee, 2023-10-07:
> It's because of the new line directive between entries. If your input file has just one line with `123`, the simulator will output `345` because it reads the input stream char by char and in that [...]

Thanks for the reply, now the result makes sense.
However, I want to know how to provide 8 bit inputs through the stream in file, because I will need at least 2 digits even if the input digits are Hex type (for example: 0xFF or 0x01).

karihepola, 2023-10-07:
> Thanks for the reply, now the result makes sense. However, I want to know how to provide 8 bit inputs through the stream in file, because I will need at least 2 digits even if the input digits are Hex type (for example: 0xFF or 0x01).

You most likely need to write some parser into the C-code that reads one line (separated by '\n') of your input file and combines the characters into an integer, [strtol](https://www.tutorialspoint.com/c_standard_library/c_function_strtol.htm) looks like a good place to start.