cpc/openasip #247: Error adding a TTA load/store instruction

claubersouza, 2023-10-30:
I'm trying to add ldw and stw instruction in LSU. I'm using adf from start.adf, from the RISC-V Tutorial tutorial.  I tried running the command "generateprocessor -o riscv-proc -t start.adf". The following error is returned:

generateprocessor -o riscv-proc -t start.adf
generateprocessor: FUGen.cc:1112: void FUGen::scheduleOperations(): Assertion `accessCycle >= 0 && "Failure likely due to mis-selected operation " "implementation"' failed.
Aborted (core dumped)

karihepola, 2023-10-30:
Hi, ProGe most likely fails, since you might have configured `ldw` to have a latency of 1 cycle. Due to the hand-shaking protocol used in the FUGen-generated load store units, load operations must have at least a two-cycle latency. I can see how this is confusing to users and the processor generator should be able to emit a more meaningful error when this happens.

May I ask what is the motivation for adding the big endian memory operations? The base architecture definition file `start.adf` already has the memory operations (base+offset little endian) correctly set for RISC-V. In addition, RISC-V custom operations must follow the RISC-V R-format (two input and one output register operand) so you would not be able to use these big endian operations anyhow.

claubersouza, 2023-11-06:
Actually, I'm trying to make an example to learn from. I took a snippet of the Fibonacci algorithm

#include<stdio.h>

int main() {

    int n = 10;
    int a = 0 , b =1;

    for (int i = 2 ; i < n ; i++) {
        int temp = a + b;
        a = b;
        b = temp;
        printf(" %d",b);
    }
    printf("\n");

    return 0;
}

I'm wanting to create an ISA for following snippet:
        int temp = a + b;
        a = b;
        b = temp;

with help from the website https://godbolt.org/. I put the algorithm in C and configured the Compiler to be gcc 12.1.0. Which ended up generating the following assembly instruction

        lw      a4,-20(s0)
        lw      a5,-24(s0)
        add     a5,a4,a5
        sw      a5,-28(s0)
        lw      a5,-24(s0)
        sw      a5,-20(s0)
        lw      a5,-28(s0)
        sw      a5,-24(s0)

Now I'm trying to create a DAG that provides a description of the algorithm I selected. The big question is how to pass a memory address to the ldw or stw instruction. OP(stw,#temp,a); I was trying to use it like this but it's not working

karihepola, 2023-11-06:
> Actually, I'm trying to make an example to learn from. I took a snippet of the Fibonacci algorithm
> 
> #include<stdio.h>
> 
> int main() {
> 
> ```
> int n = 10;
> int a = 0 , b =1;
> 
> for (int i = 2 ; i < n ; i++) {
>     int temp = a + b;
>     a = b;
>     b = temp;
>     printf(" %d",b);
> }
> printf("\n");
> 
> return 0;
> ```
> 
> }
> 
> I'm wanting to create an custom ISA for following snippet: int temp = a + b; a = b; b = temp;
> 
> with help from the website https://godbolt.org/. I put the algorithm in C and configured the Compiler to be gcc 12.1.0. Which ended up generating the following assembly instruction
> 
> ```
>     lw      a4,-20(s0)
>     lw      a5,-24(s0)
>     add     a5,a4,a5
>     sw      a5,-28(s0)
>     lw      a5,-24(s0)
>     sw      a5,-20(s0)
>     lw      a5,-28(s0)
>     sw      a5,-24(s0)
> ```
> 
> Now I'm trying to create a DAG that provides a description of the algorithm I selected. The big question is how to pass a memory address to the ldw or stw instruction. OP(stw,#temp,a); I was trying to use it like this but it's not working

Couple of comments:

1. The little endian base+offset memory operations are named `ast<n>` (store) and `ald<n>` (load) in OpenASIP. For example, `ast32` and `ald32`.
2. I assume you are trying to describe the operation pattern in osed with a DAG. The tool does not have any visibility to your C-application, hence, you must use input and output operands to describe your operation behaviour. For example, `OP(AST32, IO(1), IO(2), IO(3));`
4. I don't think the tool is able to generate a load-store unit that can chain memory operations. So I suggest you start with operation chains that don't use memory operations since those are much more simple to implement.
5. If you want to stick to RISC-V, you must make sure your custom operation can be described with 2 input and 1 output operand.

claubersouza, 2023-11-10:
Right,

I'm trying to understand how to use the DAG description and use of ast32 and al32. As I said, I have the following algorithm in C:

int temp = a + b;
a = b;
b = temp;

I'm trying to create a description of the DAG, of this algorithm:
Var c;
OP(add,IO(1),IO(2),c);
OP(ald32,c, IO(3));

One question, do you have a group on Discord to ask questions?