cpc/openasip #256: Fail in Test the processor generator and the program image generator in oa-selftest

SangYukekeaiai, 2024-04-23:
Hello There! Could you please tell me how to deal with this issue below? I saw the same issue before in #211
```
Traceback (most recent call last):
  File "/home/yy/local/bin/oa-selftest", line 597, in test_proge_and_pig
    self.assertEqual(stderr, '')
AssertionError: "Unable to copy file /tmp/tmp4ug8ilgx/pro[141 chars]l'\n" != ''
- Unable to copy file /tmp/tmp4ug8ilgx/proc_vhdl/vhdl/lsu_le.vhdl:Unable to copy '/home/yy/local/share/openasip/hdb/vhdl/fu/lsu_le.vhdl' to '/tmp/tmp4ug8ilgx/proc_vhdl/vhdl/lsu_le.vhdl'
```
I checked my install directory, the lsu_le.vhdl files are there in both openasip folder and `$HOME/local/share/openasip/hdb/vhdl/fu/` folder. The other tests are all correct.

_Originally posted by @SangYukekeaiai in https://github.com/cpc/openasip/issues/211#issuecomment-2072890208_

karihepola, 2024-04-26:
Hi,

This error indeed seems to trigger in some installations. It would be good to find out what triggers it. Have you tried running a clean installation again?

SangYukekeaiai, 2024-04-29:
> Hi,
> 
> This error indeed seems to trigger in some installations. It would be good to find out what triggers it. Have you tried running a clean installation again?

I uninstalled and installed openASIP for several times, but still didn't work well, the same issue appeared everytime.

SangYukekeaiai, 2024-04-29:
> Can you check the file permissions on /home/yy/local/share/openasip/hdb/vhdl/fu/lsu_le.vhdl' and '/tmp/tmp4ug8ilgx/proc_vhdl/vhdl/lsu_le.vhdl', and the /tmp directory? This is likely a system issue, as we cannot repro it.

`ls -al /home/yy/local/share/openasip/hdb/vhdl/fu/lsu_le.vhdl                                                               ─╯
-rw-r--r-- 1 yy yy 14325 apr 29 17:19 /home/yy/local/share/openasip/hdb/vhdl/fu/lsu_le.vhdl`
The other file under `/tmp/` does not exist. What do you suggest me to do now? do a recursive `chmod +x /tmp/`? Thx for your quick reply~

pjaaskel, 2024-04-30:
This is a strange case. Can you create files and directories under /tmp as the user you execute the test with? mkdir /tmp/bar/ && touch /tmp/bar/foo works?

SangYukekeaiai, 2024-04-30:
[image]
I tired this. It worked. But I think that is because I added `chmod -R +rwx \tmp`. But the error is still there. (I tried to uninstall and re-do the installation again but the result was the same)

pjaaskel, 2024-04-30:
OK. It's very difficult to guess what's wrong. Can you provide ssh access to the machine where this happens so we can take a look?