cpc/openasip #257: Area, Utilization, Energy Consumption estimation for the RISCV Architecture.

KulsumMohammed, 2024-04-25:
I am trying to implement matrix multiplication for RISCV for the matrix I used.
_oacc-riscv -D_DEBUG -O0 -a start.adf --output-format=bin -o matmul256_base_size4.img  matmul256_data_base.c matmul256_base.c_

I have also seen that the multiplication was successful and found the latency values.

I wanted to know the Processor resource utilization and area consumption.
In the PDF manual, to estimate the processor, the Cost/Performance Estimator provides estimates of energy consumption, die area, and maximum clock rate estimate command.

_estimate {-p [TPEF] -t [TraceDB]} ADF IDF_

To use this command, IDF is required. So, in order to generate IDF, I used the command 
_prode start.adf_

I have attached a picture of it.
[image]

Then saved the idf file as start_riscarchitecture_2.idf. Then, according to the manual, we need to use _generateprocessor_ command
_generateprocessor -i start_riscvarchitecture_2.idf -o riscv-proc_samearrayvalues_noprintf_MAvalues_idf -t start.adf._

However, I am facing errors, The error is,
**Decoder generator supports the 4-stage transport pipeline of GCU with given options. The given machine has 3 stages.
Segmentation fault (core dumped)**

I want to know how I can know the area, cost clock for RISCV architecture.
Is this the way to find the area, clock, and utilization? If so, can anyone please help with it.
 
Thank you for your time and for coming forward to help.

karihepola, 2024-04-26:
The estimator utilizes information on some standard components (standard FUs, RFs, etc.) for a specific technology node. I am not sure how well it is able to estimate customized RISC-V processors as it does not have information of the hardware complexity of the customized FUs, or of the RISC-V specifc FUs for that matter. For now, the best option is to run your design through the EDA tool flow of your choice to acquire accurate information.

As for the error you are experiencing, you can try opening your `adf` in `prode` and adding one to the delay slot parameter of the control unit, it should fix the error you are getting.

kuku464c, 2024-05-02:
Upon your suggestion, we solved the issue by changing the delay parameter of the control unit.

As for the estimating utilization, I will then use a different EDA tool,

Thank You for helping out.

KulsumMohammed, 2024-06-04:
Thank you for your time and for coming forward to help.
So instead of multiplication I wanted to try to see if I can do a custom operation for risc-V. So, I did a new operation named MAXPOOL.

[image]

[image]

I was successful with rtl generation as seen below in the riscv-proc_max_mat_O3/vhdl folder
[image]

Now I wanted to see the output, area , utilization and other by running in VIVADO.
Can you please help in how to proceed with this in vivado for analysis or any manual or pdf that might be helpful to proceed further

I am sharing all the info I have hoping that if I give a clear image of what it looks like.

Thank you for your time and for coming forward to help.

karihepola, 2024-06-04:
If you want to simply evaluate the core, you can just synthesize the `tta0.vhdl` (core entity) as the top level design to get area and clock frequency estimations from Vivado. `Section 3.11.3` in the [documentation](https://github.com/cpc/openasip/blob/main/openasip/manual/OpenASIP_manual.pdf) gives a rough example how to do this even though it is done with for AlmaIF interface but it should be directly applicable to your case if you just select the `tta0` as the top level design after importing the `riscv-proc_max_mat_O3` directory to the newly created project.