cpc/openasip #258: There's no  LSU compatibled with AlmaIFIntegrator  found in almaif.hdb

Lixiang9716, 2024-04-27:
I want to reproduce a lenet  project [https://github.com/kanishkan/lenet.git](url).However , I  found no LSU compatibled with AlmaIFIntegrator  in almaif.hdb.
[image]
How can I fix it ?

TopiLeppanen, 2024-04-29:
Hello,
The way I have been generating the AlmaIFIntegrator compatible LSUs is using the generate_lsu_32.hdb, which generates the RTL from operation snippets. So, try adding that to the --hdb-list=
You could also use the GUI to select the operation implementations from the hdb, but the CLI-version is easier, since it will automatically look for them.

Hopefully that is compatible to what Kanishkan was doing with the lenet. I haven't run that myself. It's possible that due to the version control difficulties related to hdbs, we might not now have the exact same almaif.hdb that we had back then.

pjaaskel, 2024-04-29:
@kanishkan any ideas?

kanishkan, 2024-05-01:
This issue might be related to - https://github.com/kanishkan/lenet/issues/3.. removing SIMD ops from LSU and following Topi's suggesstion should help.

Alternatively,  you can try to use one of the ADF from OpenASIP repo (<install-path>/share/tce/data/mach/).