cpc/openasip #259: How to generate the RTL?

erenyagami, 2024-04-30:
I am trying to generate the RTL and add custom instructions to improve AI kernels (convolution etc.) for RISCV .
To compile the source code for this architecture with command
_oacc-riscv -O0 -a start.adf --output-format=bin -o conv.img conv._c

To simulate the processor in RTL simulation, generate the processor RTL and its test-bench:
_generateprocessor -o riscv-proc -t start.adf_ 

After this generate the HDL packages, 
_generatebits -x riscv-proc start.adf_

Then moving to the riscv-proc folder, Compile and simulate the the VHDL source files with the
script 
_./ghdl_compile.sh_
_./ghdl_simulate.sh -r 100000_

I found the latency/cycles values and hdl_sim_stdout.txt file.

The ghdl_compile is using this
[image]

I wanted to know how I can generate the RTL or how I can proceed on it? 

Thank you for your time.

karihepola, 2024-05-02:
The RTL was generated with the command `generateprocessor -o riscv-proc -t start.adf`. This outputs the RTL in VHDL to the `riscv-proc` directory. You can find the source files for the core under the `vhdl` and `gcu_ic` subdirectories, where `tta0` is the toplevel core design.

erenyagami, 2024-05-13:
Thank You for clearing the doubt.

closed 2024-05-14 by karihepola