cpc/openasip #262: when opening prode/osed/proxim gui, Floating point exception happens.

Asplimn, 2024-05-15:
Hi Openasip Team,
When I am trying to open prode/osed/proxim gui, it says “Floating point exception“, do you have some suggestions for this kind of problem? The gcc version is 9.4.0 and the wxWidgets version is 3.1.5.
Thanks and best regards.
Renzun

pjaaskel, 2024-05-15:
Wow. No immediate ideas. Linux distribution are you running it in? Can you provide a gdb backtrace?

Asplimn, 2024-05-16:
> Wow. No immediate ideas. Linux distribution are you running it in? Can you provide a gdb backtrace?

[image]

It is the backtrace, have no idea on how to fix it.

pjaaskel, 2024-05-16:
Seems to be some sort of wx issue: https://forums.wxwidgets.org/viewtopic.php?t=36981  how did you install your wxWidgets?

Asplimn, 2024-05-16:
Hi pjaaskel,
I followed the following instructions to get/compile/install wxWidgets, please help to check is it ok. Many thanks.

wget https://github.com/wxWidgets/wxWidgets/releases/download/v3.1.5/wxWidgets-3.1.5.tar.bz2
sudo mkdir -p /usr/local/wxWidgets
tar -xvf wxWidgets-3.1.5.tar.bz2
cd wxWidgets-3.1.5/
./configure --with-regex=builtin --with-gtk --enable-unicode --disable-shared --prefix=/usr/local/wxWidgets
sudo make && make install
cd /etc/ld.so.conf.d/
sudo vim wxWidgets.conf and add content "/usr/local/lib" to it
sudo ldconfig
sudo vim /etc/profile and add content "# wxWidgets
                                                                export WXPATH=/usr/local/wxWidgets/
                                                                export PATH=$WXPATH/bin:$PATH" to it
source /etc/profile
wx-config --version

pjaaskel, 2024-05-16:
Are you able to run any wxWidget application? I think they have some test applications in there. Why did you install it from sources and not from your distribution's packages?

Asplimn, 2024-05-16:
Hi pjaaskel,
It's my fault. The default wxWidgets of my distribution's packages is 3.2.4, cause the openasip configure file says the supported version is 2.6/2.8/2.9/3.0/3.1, so I guess 3.2 may not be ok for this. According to your advice, I add 3.2.* to the configure.ac file and rebuilt/reinstall openasip, and it works fine. Thanks so much for your kind help.
Best regards,
Renzun

pjaaskel, 2024-05-16:
Good to hear 3.2 also works.

closed 2024-05-16 by pjaaskel