cpc/openasip #271: Issue in generating the VHDL for the processor using Processor Generator (ProGe)

Jedrek09, 2024-08-29:
Hello. I was having a problem in generating the VHDL for the processor using Processor Generator (ProGe). I did the  selection of implementations for register files, immediate units, and function units. Due to the selection window not being very informative about the different implementations, I opted for automatic selection instead.I also enabled bus tracing from the Implementation-dialog's IC / Decoder Plugin tab and clicked "Save IDF" and named the file custom.idf.  I then clicked "Generate Processor," created a directory named proge-output, and attempted to generate the processor. I encountered an issue during generating the VHDL for the processor using Processor Generator (ProGe). However, the dialogue box disappeared, and the terminal displayed an error. I also tried the alternate method using terminal commands, but I encountered errors there as well. Could you please guide me on how to resolve this issue?

karihepola, 2024-08-29:
Hi,

Could you please paste the error you get when calling

`generate_processor -i custom.idf -o proge-output custom.adf`

in the terminal?

Thanks

Jedrek09, 2024-08-29:
This is the error I get after running generateprocessor -t -i custom.idf -o proge-output custom.adf

FU entry 1 does not have an implementation in HDB /home/salman/openasip-devel/openasip/tce_tutorials/tce_tour/tour.hdb.
Segmentation fault (core dumped)

karihepola, 2024-08-29:
Did you double-check that the entry in the HDB is as described in the tutorial? Were you able to add the implementation of the function unit successfully? Additionally, the IDF is an XML file that you can open with the text editor of your choice to inspect it. Does it point to a valid function unit entry in the HDB?

Jedrek09, 2024-08-29:
Yes I rechecked the entry in the HDB as described in the tutorial. For the function unit implementation I used the  automatic selection and was successful. After opening the IDF file in text editor I saw it points to a valid function unit entry in the HDB but is still showing error

pjaaskel, 2024-08-29:
Can you send the files for us to test?

karihepola, 2024-08-29:
Could you also add the adf and hdb?

Jedrek09, 2024-08-29:
[custom.adf.txt](https://github.com/user-attachments/files/16803364/custom.adf.txt)
[tour.hdb.txt](https://github.com/user-attachments/files/16803367/tour.hdb.txt)

Here are the required file you asked for.

karihepola, 2024-09-01:
Sorry for the delay.

I cannot open the hdb file as it seems to be somehow corrupted. Can you add it as a zip without adding any extra extensions (.txt) to the file name.

Thanks

Jedrek09, 2024-09-01:
[custom.adf.zip](https://github.com/user-attachments/files/16829858/custom.adf.zip)
[custom.idf.zip](https://github.com/user-attachments/files/16829859/custom.idf.zip)
[tour.hdb.zip](https://github.com/user-attachments/files/16829860/tour.hdb.zip)

Here are the required files you asked for.

karihepola, 2024-09-02:
Still cannot open it. Are you able to open the hdb file with _hdbeditor_ or an sqlite tool such _sqlitebrowser_?

Jedrek09, 2024-09-02:
Yes by installing hbdeditor and sqlitebrowser through terminal command I was able to open the hdb file and view it

karihepola, 2024-09-02:
I think the file is formatted as text (due to the .txt extension) at some point, which corrupts it. Can you upload the original file in the original format and check that it works by downloading it yourself after uploading it to github?

Thanks

karihepola, 2024-09-02:
Thanks, now it works.

If you inspect the idf, you can see that the reflecter function unit is mapped to fu entry 2 in the tour hdb, but the function entry 2 does not have any implementation defined in the hdb. I think you have accidentally added two empty definitions for this function unit in the hdb, which is causing issues. If you modify the idf by hand to reference fu entry 3, it will work, since that fu entry has an implementation defined.  It will first give you an error though about missing load port definition, but it should work once you add that in the hdb by navigating (as in the tour instructions):

Function units -> reflect32(1) reflect8(1) -> Modify -> t1data p1... -> Load port

pjaaskel, 2024-09-03:
Should not segfault though, keeping open.

Jedrek09, 2024-09-04:
I am still facing error in missing load port definition part. Could you please explain the later part a bit elaborately like about the hdb file portion

karihepola, 2024-09-05:
You need to revisit this part of the tutorial.

[image]

Jedrek09, 2024-09-05:
Thanks I was able to solve the issue of load port definition. I was able to generate instruction memory bit image using Program Image Generator and was able to visualize the count of instructions by counting the lines in that file. However while simulation and verification after going to the proge-output directory I faced issue in compiling and simulating the testbench. After running ./ghdl_compile.sh
./ghdl_simulate.sh

I get the error 
./ghdl_compile.sh: line 58: ghdl: command not found
./ghdl_simulate.sh: line 98: ghdl: command not found

after using modelsim 
./modsim_compile.sh
./modsim_simulate.sh

I get the error
/modsim_compile.sh: line 51: vlib: command not found
./modsim_compile.sh: line 52: vmap: command not found
./modsim_compile.sh: line 60: vcom: command not found
./modsim_simulate.sh: line 102: vsim: command not found

Can you help me to solve this

pjaaskel, 2024-09-05:
Yes, you do.

Jedrek09, 2024-09-05:
After installing by manually using terminal command sudo apt install ghd1 I got the error

 ./ghdl_simulate.sh: line 99:  5210 Killed                  ghdl -r ${std_version} --workdir=work --ieee=synopsys ${tb_entity} --stop-time=${runtime}ns --ieee-asserts=disable-at-0 ${generic_list}

I made sure  the bit widths of data and instruction address spaces were set to 15 bits

pjaaskel, 2024-09-05:
Did you remember to generate the VHDL again after doing so? This crash typically happens when you run out of memory due to the too large SRAM array created.

Jedrek09, 2024-09-05:
Yes I have generated the VHDL again after doing so. Its still showing the same error

pjaaskel, 2024-09-06:
How much memory does your PC have? Does your program simulate correctly in oasim?

Jedrek09, 2024-09-06:
My PC have installed physical memory (RAM) 8.00 GB. In ubuntu I set the RAM to 4096 MB. 

My oasim test showed okay. (Image attached)
[image]

pjaaskel, 2024-09-09:
That's the standard smoke test. Which program you are loading to the ghdl simulation? Does it run ok in oasim?

Jedrek09, 2024-09-10:
I was able to generate instruction memory bit image using Program Image Generator. While  compiling and simulating the testbench with GHDL I faced an error. Yes, it runs okay in oasim

pjaaskel, 2024-09-11:
Interesting. Can you send the program source, TPEF and ADF you used?

Jedrek09, 2024-09-11:
Please check the attached files

[reflect.zip](https://github.com/user-attachments/files/16959652/reflect.zip)
[crc.zip](https://github.com/user-attachments/files/16959761/crc.zip)
[custom.zip](https://github.com/user-attachments/files/16959759/custom.zip)

karihepola, 2024-09-14:
I cannot generate the bit images because the adf and the tpef do not match (you used different architecture to compile the program). Could you provide (one) zip file that is the rtl folder with the bit images, scripts and vhdl files included. Thanks

Additionally, which version of ghdl are you using? I think the one you get from the default system packages is _very_ old

Jedrek09, 2024-09-14:
Please find the attached file
[proge-output.zip](https://github.com/user-attachments/files/17003695/proge-output.zip)

My ghdl version shows the following

GHDL 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
 Compiled with GNAT Version: 10.3.1 20211117
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

karihepola, 2024-09-16:
Hi,

It seems that you still have the data memory address space defined as 32bit, which allocates lots of memory and causes the simulation to crash. You can see this from the file `vhdl/tta0_params_pkg.vhdl` `  constant fu_lsu_dataw : integer := 32;
`. Make sure that you have this defined to a reasonable value, like 24b and generate the processor RTL again

Jedrek09, 2024-09-16:
Can you elaborate the solution a bit? Should I start the whole process from the beginning?

karihepola, 2024-09-16:
You need to restrict the data address space by first opening the .adf in `prode` and then:

Edit -> Adress Spaces... -> data -> Edit... -> Bits -> 24

After this, recompile the program using `oacc` and then regenerate the RTL with `generateprocessor`

Jedrek09, 2024-09-16:
Recompiling the program using oacc! does that mean  Creating the custom operation again?

Jedrek09, 2024-09-16:
I edited the data bits to 24 and did the entire process from the scratch. Still was facing the issue :

./ghdl_simulate.sh: line 99:  5209 Killed                  ghdl -r ${std_version} --workdir=work --ieee=synopsys ${tb_entity} --stop-time=${runtime}ns --ieee-asserts=disable-at-0 ${generic_list}
salman@linux:~/openasip-devel/opena

karihepola, 2024-09-17:
Could you send me all the necessary test files (.adf, .hdb, .opp, .c and .idf) that you use in (one) zip file? It seems that the address width was correct the first time around, and there is something else that is causing the simulator to allocate memory. It is difficult to debug this further directly from RTL alone.

Jedrek09, 2024-10-01:
Please kindly see the whole file attached. It contains all the documents including the ones you asked for. Let me know if you need further informations. 

[tce_tour.zip](https://github.com/user-attachments/files/17199727/tce_tour.zip)

karihepola, 2024-10-11:
Thank you for sharing the rest of the source materials.

I gave it a spin, and I noticed that overestimated the memory sizes ghdl can handle for simulation in my earlier message. I reduced both the memory (data and instructions) to 15 bits (also mentioned it the tour instructions) and the simulation was able to run successfully.

If you have further issues with the tutorial, you can use the tce tour system test as reference, since it includes all the files and scripts to run the tour:

[https://github.com/cpc/openasip/tree/main/testsuite/systemtest/codesign/tce_tour](https://github.com/cpc/openasip/tree/main/testsuite/systemtest/codesign/tce_tour)

Jedrek09, 2024-10-13:
I would like to mention that I did convert it to 15 bits at that time

Jedrek09, 2024-10-14:
Can you explain a bit precisely how I can use the reference tce tour like which file should I take into consideration i.e how to use that

Jedrek09, 2024-10-14:
Should I start the tutorial from scratch since a lot of modification has been done and I have lost the track?
Will it affect the output if I do so?
Should I delete the entiire tce_tour file and start from the beginning?
I am having issues while getting the output for custom operation now

Jedrek09, 2024-10-14:
Can you kindly send the tce_tour zip file of yours for me to test and generate the output?

Jedrek09, 2024-10-14:
Thank you I was able to simulate the processor VHDL and got the output 

/usr/bin/ghdl-mcode:info: simulation stopped by --stop-time @52390ns

and produced file “bus.dump"

whenever I do the verification I get the output printed. Can you help me in this case

closed 2026-01-29 by pjaaskel